gdombiak/OctoPod #332: Implementation of new Tradfri outlet plugin

ScharV, 2020-05-26:
Hey @gdombiak there is a new plugin called Ikea tradfri which allows you to switch on and off your printers ikea outlet. Would be nice to do this via octopod too. 
Plugin: https://github.com/ralmn/OctoPrint-Ikea-tradfri

Best regards ScharV

gdombiak, 2020-05-26:
Nice new feature. Do you know how popular is this plugin? This plugin has its own unique API so adding support for it would be more involved. The other power plugins work very similar (since they were done by the same developer) so adding support for them was not much work.

I'm with limited time these days so trying to optimize it. :)

Thanks,
Gaston

ScharV, 2020-05-26:
Thanks for your fast reply💪🏼
Greetings Marv

ralmn, 2020-06-21:
Hi,

I'm the creator of Tradfri plugin, I'm also interesting with this feature. I don't have time to see how to implement this feature in your app. But if you want my help to add endpoint in the API I can do that ;)

gdombiak, 2020-06-22:
Hey @ralmn, nice meeting you!! It would be convenient to have the following APIs:

1. API to get list of devices and their statuses if possible
2. API to be able to turn on/off and pass name of device. API returns success code or some error code if operation failed.

Thanks,
Gaston

ralmn, 2020-07-16:
Sorry for the delay. The ikea plugin handle only one plug. So the two endpoint are : 
* `/plugin/ikea_tradfri/state` with method `GET` would return json : `{ "state": true }` or `{ "state": false }`
* `/plugin/ikea_tradfri/state` with method `POST` and body in JSON (header `Content-Type: application/json`) `{ "state": true }` or `{ "state": false }` :  would set the plug state and return json : `{ "state": true }` or `{ "state": false }`