gdombiak/OctoPod #333: Octopod Plugin + Octopod IOS: Get 10 Push-Messages for every status (25%, 50%, 75%)

Gosarianer, 2020-06-01:
Hi...
I'm using Octoprint with Octopod Plugin and Octopod-Client on Iphone XR and Apple Watch.
I only got one printer "Anycubic I3 Mega" connected. Every status update (25%, 50%, 75%) fires 10 push notifications, that immediatly appears on Iphone and Apple Watch, which is annoying. The completion message finally comes twice. One with printer name in head of message, the last without a printer name. 

Look at the screen shot: 

[image]

The configuration page in plugin:

[image]

Is this just a configuration problem?

Thank you...

Regards,
Gosarianer

gdombiak, 2020-06-01:
Ouch! The only reason I can think of for getting multiple notifications is that in OctoPrint in the config.yaml file somehow the OctoPod app got registered multiple times. Open that file and under plugins -> octopod -> tokens you can delete all the entries that appear there that start with a -.  Restart OctoPrint and open again the OctoPod app on your phone. Make sure to have a backup of config.yaml before making any changes.

If you figure out the steps that produced this situation let me know and I will see if there is anything I can do to prevent it.

There is already a ticket that will let you see all registered devices (i.e. tokens) with this plugin so you can delete then from the UI instead of manually modifying this file.

Thanks,
Gaston

gdombiak, 2020-10-05:
Hi @Gosarianer, I will close this ticket since release 0.2.7 of OctoPod plugin for OctoPrint now supports https://github.com/gdombiak/OctoPrint-OctoPod/issues/43 which lets you control and delete duplicates.

closed 2020-10-05 by gdombiak