gdombiak/OctoPod #334: [FR] Display already printed files in a different color

DominikPalo, 2020-06-04:
It would be useful to use a different color for already printed or cancelled/failed files in the files explorer (like in OctoPrint UI, where printed files use green text and cancelled/failed files use red text).

See screenshots:
[image]

[image]

gdombiak, 2020-06-04:
Love this improvement. Will see what information does OctoPrint expose so I can mimic this behavior.

Thanks,
Gaston

gdombiak, 2020-06-05:
[image]

commit a23a69d3 (2020-06-05): Added color to list of files to indicate if last print was successful or not. Same as OctoPrint UI does

Closes #334
closed 2020-06-05 by gdombiak