gdombiak/OctoPod #335: OctoPod shows different file sizes

DominikPalo, 2020-06-04:
As you can see on the attached screenshot, the sizes of the files displayed in OctoPrint are different when compared to sizes in OctoPtint UI. Not a big deal, I'm just curious. Maybe it is due to a different formula used for the conversion from bytes to MB (SI conversion, where 1 MB = 10^6 B vs binary conversion, where 1 MB = 2^20 B)?

[image]

gdombiak, 2020-06-04:
Wow. That is SUPER weird! :) Could you upload one of those gcode so I can reproduce it locally and fix it?

Thanks,
Gaston

DominikPalo, 2020-06-04:
Of course, here is it: [F9_Land_Gear_Mount.gcode.zip](https://github.com/gdombiak/OctoPod/files/4731838/F9_Land_Gear_Mount.gcode.zip)

commit 1a3d02f9 (2020-06-05): Fixed incorrect file size

Closes #335
closed 2020-06-05 by gdombiak