gdombiak/OctoPod #336: iOS app stops printing when quit

barnsFromOz, 2020-06-04:


gdombiak, 2020-06-04:
Hi @barnsFromOz, I cannot think for a reason why closing or quitting the app would stop printing. Something else must be going on for this to happen. When you close or quit the app, all that happens is that the network connection for reading status gets closed. 

Assuming you are printing from OctoPrint (and not from SD card), the OctoPrint server is the one telling the printer what to print. I can only imagine of a malicious plugin that could listen to a websocket connection getting closed and deciding to tell OctoPrint to stop printing. But this would be very weird.

In summary, no idea how this could be happening. OctoPod cannot be causing this behavior on its own.

Regards,
Gaston

barnsFromOz, 2020-06-05:
Hi @ Gaston 
Ok. That makes sense... all I can do is test it again. I'll try again tomorrow (this time in a small job :-) ) and let you know. Cheers, Barns

barnsFromOz, 2020-06-07:
Hi @gdombiak 
Not good news, and no ideas either. I started printing one of the TestBed Levelling jobs, 15m run. Two tests, one before printing started, the second while the extrude head was moving around the platen. Backgd: I have an Octoprint server (an a RasPi 3) and was attached using OctoPrint on the browser of a Mac laptop. the 3D printer is a Creality CR-10 v2.
Test 1: started OctoPod up, and sent the gcode file. Everything proceeds as normal. Then I chose Disconnect  while extruder was heating up. Nothing changed at the printer, but when the head reached 200°C it stopped. Both bed and head stay at this their target temps. 
No printing started. I chose 'Cooldown' on the CR10 screen. After that, I was able to connect again with OctoPod, BUT the Print option stayed grey. 
I restarted the App, and things appeared to be normal. 

Test 2: Ran the app, it auto connected to the server as normal. 
a) Sent the same gcode file to printer. Waited for head to start moving (screengrab1)
b) On OctoPod, chose Disconnect (Sure? Y etc)
c) the CR10 continued for a second, as tho it were finishing the current command, then stops. No head withdrawal, just stops. (grab2)
d) the temps stayed at Target settings, as before. 
Looking on the laptop the OctoPrint screen now says (grab3)
State offline, and Terminal window shows the last lines sent (grab4) as:
   Send: N1147 G1 F1200 X71.149 Y202.954 E144.17012*2
   Recv: ok
   Send: N1148 G0 F6000 X71.149 Y203.52*120
   Connection closed, closing down monitor
   Changing monitoring state from "Printing" to "Offline"

I cant actually see a command which says STOP, unless N1148 does that...but it looks like a print command. 

I can forward the 4 screen grabs if you would like, but they only confirm the descriptions above.  I wonder if I have provided enough to diagnose the problem? 

Cheers,  Barns

gdombiak, 2020-06-07:
Hi Barns,

I think I know what is going on. The **disconnect** button disconnects OctoPrint from your printer. The same way **connect** button will connect OctoPrint to your printer. If you are using an 8-bit board then upon connection the board will restart. If you are printing from OctoPrint and click disconnect then the printer will stop printing.

Under Settings->Prompts you can configure OctoPod to show a confirmation dialog when connecting or disconnecting to prevent unwanted issues like what you are sharing.

If you do not want to disconnect OctoPrint from the printer and you want to close OctoPod then just turn off the phone, or switch to another app or if you prefer to kill the app then any of those options will NOT affect any running job.

Hope this helps. Let me know if you have further questions.

Regards,
Gaston

barnsFromOz, 2020-06-07:
Ok, thanks for your quick reply. 

May I ask two questions: 
1. I think you are saying that printing from OctoPrint on my laptop is different from printing from the OctoPi server, mainly cos the OPi server (based on the RasPi v3 board) is an 8bit board? And the answer is  don't choose Disconnect using OctoPod in those circumstances? Clear advice, I get it...
2. Is there anyway to easily restart a print job which is part-printed?  Like the second one I have described above?

Cheers,  Barns

ariforu, 2020-06-25:
@barnsFromOz I think @gdombiak is referring to your printer controller board (that came with the printer) not your Raspberry Pi that you added. If your printer controller board is 8 bit then it'll restart itself when you make OctoPrint server (running on your Raspberry PI) makes a connection to it. You do not need to tap "Disconnect" if you intend to just quit OctoPod and don't want Octoprint Server to disconnect from the controller. Printing from your laptop (e.g. Web browser) with OctoPrint UI is not different from printing from your phone with Octopod. Both your browser and OctoPod are "clients" to Octo Print server running in your Raspberry PI. @gdombiak please feel free to correct me.

gdombiak, 2020-06-25:
@ariforu you got it 100% right! :)

gdombiak, 2020-10-05:
Closing the ticket. We can reopen it if there is something to fix in OctoPod.

Thanks,
Gaston

closed 2020-10-05 by gdombiak