gdombiak/OctoPod #349: [FR] Fixed width font in Terminal view

jpflouret, 2020-07-01:
It would be nice if the terminal view used a fixed width font. Any kind of formatting of tables gets lost/is hard to read.

gdombiak, 2020-08-16:
Hi @jpflouret, could you give me an example where I can see the problem? This will help me confirm that the fix solved what you requested.

Thanks,
Gaston

jpflouret, 2020-08-31:
[image]

Notice how the font in the terminal window is not fixed width and thus the repeating text doesn’t align. This is especially bad when looking at a table like the result of M420 V in Marlin.

gdombiak, 2020-08-31:
Gotcha! That helps a lot! Will see what I can do for the next release.

Thanks,
Gaston

gdombiak, 2020-09-12:
[image]

commit a94ae7e1 (2020-09-12): Use monospace digits in terminal so it looks better. Added missing color to gcode field

Closes #349
closed 2020-09-12 by gdombiak