gdombiak/OctoPod #350: Octopod external network

DemiDieuAscar, 2020-07-01:
Hello,
I cannot access my printer outside my local network, but I do receive notifications on the application when the printing is finished.
is it possible to connect in 4G to see the progress of printing?

jessearmstrong, 2020-07-01:
OctoPod is only able to connect to Octoprint instances on a local network. You can set up a VPN tunnel to your home network and when connected it will work just like if you were at home. I do this and it works great.

gdombiak, 2020-07-01:
@jessearmstrong is spot on. You can follow [this guide](https://github.com/gdombiak/OctoPod/wiki/Using-OpenVPN-for-Remote-Access) to learn how to setup your VPN. For a less secure option but easier to setup you can use a reverse proxy with authentication and port forwarding.

DemiDieuAscar, 2020-07-02:
Hello,
Thank you for your answers @jessearmstrong @gdombiak

gdombiak, 2020-10-05:
Closing this ticket unless there is something to fix in OctoPod.

Thanks,
Gaston

closed 2020-10-05 by gdombiak