gdombiak/OctoPod #351: Apple TV app not working

lupleime, 2020-07-05:
When I download the apple TV it just says retrieving printers information

sysadmininator, 2020-08-02:
Same here. App worked great up until a month or two ago. I want to say it stopped right around the time the latest tvOS update was pushed, but that could be purely coincidental. iOS version continues to run without issue. If anyone can point me in the direction of gathering logs I’d be happy to track mine down and post. Thanks!

gdombiak, 2020-09-19:
I just saw this after I updated to the latest tvOS version. I tried to troubleshoot but it when I tried to debug it the problem got resolved. :(

I will push a new version of the app now and let me know if this helped get things unstuck. The error is still there though and might resurface some other time. I hope I will get to it before then.

sysadmininator, 2020-09-19:
@gdombiak thanks for the reply on this. I actually ended up resolving this one a few days after my post. Sorry I forgot to come back and report my success. For me what ended up working was to use the ‘Reset iCloud data from printers’ option in the Settings > Devices area of the app. I can’t remember if I turned off the sync option before I did this or not, or if I did any other force quitting of the app on iOS or tvOS before things started working again. Might have also been a new version that got pushed, but I want to say that I checked for updates around the time I was having issues and I was current on both devices.

gdombiak, 2020-09-20:
That's a good hint. I will see if a change in the database structure of the iOS app could cause this issue on the Apple TV app. My case happened around the same time so will test that case and see if I can reproduce it.

Thanks,
Gaston

sysadmininator, 2020-09-20:
Quick update as this just happened again a few minutes ago. iOS and tvOS updated to 14 in the last couple of days. iOS has been solid I hadn’t even tried tvOS until tonight. Sure enough “retrieving printers information” screen. I force closed the app on tvOS, popped over to iOS and reset ‘iCloud data from printers’, opened octopod on tvOS again and everything was back to working.

I’m not sure if or when it’ll happen again, but if it’s helpful to grab any logs if there is a next time, I’m happy to help. Otherwise, that rest method appears to do the trick.

srisinger, 2021-03-03:
I just installed the Apple TV app, version 3.6, and I'm experiencing the same issue. iPhone version 3.11. I even renamed my printer on my phone to force changes to be pushed. Force closed both apps. Unfortunately, I'm just not getting my printer to load on the TV. 

Thoughts? Any logs I can pull for you on this?

jatkins11, 2023-09-23:
Not sure if this is tied to the same cause or something else. If it is something else I will open a separate issue. I recently installed the Octopod tvOS app and it just says “Retrieving printers information”. Octopod on iPhone works fine without issues. 
I had tried a few pointers listed in other reported issues but none of them have worked.
- updated to latest version of everything: tvOS, octopod app, octopod plugin, octoprint.
- Ensured iPhone and Apple TV are setup to sync iCloud data.
- Removed printers and readded.
- Went into settings and changed the name to save it again.
- Deleted apps and reinstalled.

if anyone has any tips on how to resolve id appreciate it. 
[image]