gdombiak/OctoPod #353: OctoPod Swag?

amodernstitch, 2020-07-09:
I am sorry if this is not the right spot to post this but couldn’t find another appropriate space. I am just getting started in 3D printing and found your App for all the Apple Things. 
Now to the point of this post... I Love the design of your App Icon. I would proudly have it as a sticker on my MacBook or on a Tshirt. Anyway we can support you through buying swag?

gdombiak, 2020-07-09:
Thanks so much @amodernstitch for the kind post and offer. It's certainly an encouragement to keep giving to this awesome project! It's an awesome idea having stickers or TShirts. It'd be great to get help kicking this off. Let me know if this is something you'd be interested in helping with. 

What is available now is [sponsoring](https://github.com/gdombiak/OctoPod) this project. Look on the right side to find the sponsor button. It is not the same as your cool idea though.

Thanks,
Gaston

T3chC0wb0y, 2020-07-20:
If someone wants to send me the Octopod logo in a high quality format, (.pdf .png etc..) I can see what it would cost to have some stickers printed.

amodernstitch, 2020-07-20:
It looks like TeeSpring is a go to place for a lot of on demand printed shirts. I am happy to set it up. I will have to look into how the financial side of this works and how the proceeds can get to you. 
I would need the amazing artwork to get started.

gdombiak, 2020-08-16:
@T3chC0wb0y, is this 1024x1024 good enough? [image]

I can also share an SVG with no background color.

Thanks guys!!!
Gaston