gdombiak/OctoPod #354: Camera view upside down on iPad

juchabox, 2020-07-13:
Hi I use Octopod on iPhone and iPad with the same Octoprint server on Raspberry Pi connected to a single printer.
On the OctoPrint UI I flipped the camera view as it is looking to the print bed upside down. The view on Octoprint UI is correct. It is also correct on OctoPod on my iPhone. But on my iPad, OctoPod show my camera view upside down.

IPad Pro 11in
Ipad OS 14.0
OctoPrint 1.4.0

juchabox, 2020-07-13:
I close the app and restarted it.
The view is in the correct direction now.
Sorry about the noise.

juchabox, 2020-07-13:
I closed the app and restarted it.
The view is in the correct direction now.
Sorry about the noise.

closed 2020-07-13 by juchabox

gdombiak, 2020-07-13:
I was going to reply that it is the same code so it was weird. Thanks for updating the ticket with the resolution.

Gaston