gdombiak/OctoPod #355: Bed Leveling From Within the App

T3chC0wb0y, 2020-07-20:
I have tried all the Octoprint compatible apps for iPhone, and this one blows the others out of the water.  That being said, I have yet to find a way to automate the five point bed leveling locations through the app.  There are plugins for Octoprint that make this possible through the browser, but none of them currently port over to the app.  Am I doing something wrong or would this be a new enhancement?

[image]

gdombiak, 2020-10-05:
Hi @T3chC0wb0y, which plugin are you using that would like to control from OctoPod? Once you pointed me to the right plugin, I will check feasibility of supporting that plugin.

Thanks,
Gaston

bastienstefani, 2021-09-16:
Hello, just in case you didn't received already an answer for this, personnaly i use https://github.com/jneilliii/OctoPrint-BedLevelVisualizer which provide a clear map of the bed levelling. I would love to see this in the app !