gdombiak/OctoPod #357: Cant add or update printer - save button greyed out and disabled

dan-glass, 2020-07-31:
Since upgrading to the latest iOS version 3.5 the "save" button on the "Printer Details" screen remains greyed out no matter what is entered:
 - photo or api  key string
 - http / https
 - enter port number in url
 - ip or host name
 - username valid or missing

The printers entered before upgrading work, but they cannot be modified.

(NOTE: The "scan" function doesn't find anything, so I was not able to test whether that would cause the save button to be enabled)

dan-glass, 2020-07-31:
Deleting app and settings, and reinstalling fixes issue.

kylerw, 2021-08-12:
I am experiencing the same thing

gdombiak, 2021-08-12:
Hey @kylerw ,

Could you share screenshots so I may see what is going on? BTW, after you completed required fields (top 3), let me know if save button is enabled after you tap on any other part of the screen.

Thanks,
Gaston

Xalior, 2021-08-29:
I too am getting this, and removing and reinstalling the app does not let me add a new printer.  I also cannot delete my old printer, or disable iCloud syncing, meaning new installs of the app contain my old printer.  To be clear I don't want to remove my old printer, I just want to add a second new printer -- but tried to remove the old one as a debugging step.  Have tried to add via App API, OctoPrintEverywhere, and deprecated Global API, on two different iPhones, and the problem is always the same - save button never activated, so could not create new printer.

gdombiak, 2021-08-29:
Hi @Xalior, sorry to hear the save button is putting up a fight. No need to delete a printer or reinstall the app to make this work. Could you do a simple test to see if this works? Once you completed all required fields (make sure you have a name, hostname and api key if you are using app key or global api key) then tap with the finger on any other field so your last field loses focus. Let me know if save button is enabled after this.

It would help me if you can tell me the exact steps you followed (maybe take a screenshot for each step) so I can reproduce this problem and see how I can fix it.

Thanks,
Gaston

Xalior, 2021-08-29:
Pick Local Auth:
[image]
Pick my Printer from the "scan" button:
[image]
Enter username, have even tried creating different user in octoprint:
[image]
Enter password, same credentials work fron from browser:
[image]

Have tried clicking out of password box, onto other widgets at this point, tried toggling the switches for camera, tried tapping into the empty space in the UI layout, tried tapping the disabled save button, nothing progresses out of this screen except "Cancel".

Xalior, 2021-08-29:
And, to confirm, the printer connection dialog box does contain correct, sane, details from the Scan...
[image]

gdombiak, 2021-08-29:
Hi @Xalior,

Thanks for sharing this great info. In this case, I see that you are using "Application Key" for adding a printer. The save button is disabled until you clicked on the **Request** button at the bottom and from OctoPrint you approved the request to give OctoPod an application Key.

However, in your screenshots I see that the **Request** button is also disabled. Since you completed the name of the printer and the URL address, the only reason to have that button disabled is that the app is locked. In OctoPod, could you go to Settings -> Security and make sure that **Lock read-only mode** is disabled?

Gaston

Xalior, 2021-08-29:
And that has sorted it! 

I can delete old printers too now (which was another “bug” which was bugging me)

I never turned that on (to my memory) and the “lock bit” has synced across all my devices, via iCloud I presume (I do have sync enabled) and lead to a very frustrating week of trying various things, uninstalling and reinstalling, etc!  

Thanks for your help!

gdombiak, 2021-08-29:
Excellent!! This tells me that I should show some indicator that the app is locked!!! The lock is handy when you are using Apple TV and want to lock print controls or if you handle the phone to someone and you do not want them to mess with your print job.

Xalior, 2021-08-29:
I concur that it’s a good feature and — while I don’t deny turning it on, I never turned it on intentionally, or consciously — but I’m not saying you have a bug there, but I am wondering when I accidentally engaged that...

The fact I didn’t know it was on, however, has been “the issue” for me for a whole week, and I’ve definitely “wasted" a lot of time trying to debug the situation before contacting you.  

I notice you’ve put a lot of new iconography in the latest build (sorting the printer list, thumbnails in the camera view, etc.) so maybe a new padlock in the printer list on the lower left (opposite side to the sort icon) when the app is in R/O mode would be enough?

-Dx

gdombiak, 2021-08-29:
How about this idea? I replaced the + button and the Save button for a lock 

[image]

[image]

Xalior, 2021-08-29:
Much better idea ;-)

gdombiak, 2021-08-29:
Excellent. Will go with this then for the next release. Thanks @Xalior for the feedback

commit 3dd7a185 (2021-08-29): Show lock in + button or save button so user knows why they cannot save changes, add new printers or delete existing ones.

Closes #357
closed 2021-08-29 by gdombiak