gdombiak/OctoPod #359: iOS 14 beta error

webpty, 2020-08-11:
I've recently updated my iPhone 7+ to iOS 14 beta, and i get an error. That being said. it's not a production version so I dont pretend everything is going to be working flawlessly. I thought i'd let you know, though...
I currently get this message: 
```"the internet connection appears to be offline"```

Clicking on the  camera stream link does open the camera link, but I have no control over the printer.

My iPad which is on the latest iOS 13 production version still works beautifully.

[image]

gdombiak, 2020-08-11:
Hi @webpty 

Thanks for bringing this up to my attention. I haven't tested the app with the new iOS 14 yet. I will see if I can reproduce this issue. BTW, does it work fine if you are not using VPN?

Thanks,
Gaston

webpty, 2020-08-11:
Hi @gdombiak,
I actually had not checked that, but removing it did not make a difference.
Thanks!

webpty, 2020-08-11:
Hi @gdombiak 
I found out it was just a permission issue.
As it is a permission automatically granted previously, and not on iOS13, it makes sense that Octopod doesn't ask for it.
Permission found in `Settings > privacy > local network`

[image]

[image]

jessearmstrong, 2020-08-11:
You mean you don't want Facebook to be able to access your local network!?

</s>

webpty, 2020-08-11:
I think the wording i've seen on the dialogs is something like 
> Do you want to allow FB to look for devices on your network?

Thanks, but no thanks

gdombiak, 2020-08-11:
Thanks guys for tracing OctoPod more than Facebook app. hahahaha :D

gdombiak, 2020-08-11:
Great finding @webpty. You just saved me some time researching what was going on. Seems to be a very easy fix.

gdombiak, 2020-08-16:
[image]

Fixed and tested on iOS 14 Beta 4. 

Thanks for reporting this!
Gaston

closed 2020-08-16 by gdombiak

gdombiak, 2020-08-17:
@webpty & @jessearmstrong, I created a new beta version for the next release that includes this fix and the Siri one that Jesse reported and many other things I worked over the weekend. Let me know if you guys are interested in being part of the beta testing team. Send me an email to gdombiak at gmail.

Thanks,
Gaston

jessearmstrong, 2020-08-17:
I’d be happy to help beta test. Anything you need!

webpty, 2020-08-17:
@gdombiak Of course! count me in!

gdombiak, 2020-08-18:
Excellent guys! Send me an email to the address I shared above. Once Apple approved the new beta I will share the link with you. You will need to install TestFlight app if you do not have it already.

Thanks,
Gaston

Neokamikaze, 2021-04-26:
Hello, I have the same problem (my phone is Iphone 12 on iOS 14.4.2). Access to the API from a Chrome interface is very fast but from the application it is impossible to retrieve the print data.
Thanks

commit 458387f0 (2020-08-16): iOS 14 changes. Local network and Bonjour permissions are now supported.

Closes #359