gdombiak/OctoPod #360: Support for webcam stream on different port?

gee4vee, 2020-08-13:
I have my mjpg-streamer running on a non-default port and configured OctoPrint accordingly. When I'm on the local network, I can see the webcam feed just fine on OctoPod and in the browser. However when I try to use OctoPod remotely I don't get the webcam feed. I can see the rest of the print information is being updated. 

I forwarded my OctoPod and mjpg-streamer ports to my rpi in my router configuration. I set up the printer in OctoPod to use my dynamic DNS address so that I can use it remotely.

Does OctoPod support a non-default port for the webcam stream? Or could it be my rpi is struggling to send the feed over the internet?

gdombiak, 2020-08-13:
Hola Gabriel,

OctoPod should be able to connect to any machine at any port as long as connectivity setup is fine. You seem to be doing port forwarding so double check that you added a port forward for the webcam. When OctoPod fails to connect to the webcam, you should see the link below the error message. You can click on the link and that will open safari so you can see and quickly test connectivity from safari.

Hope this helps,
Gaston

gee4vee, 2020-08-19:
OctoPod doesn't give any error message about the feed, so it appears that the rpi can't handle sending the feed over the internet. Thanks for the help anyway.

closed 2020-08-19 by gee4vee

gdombiak, 2020-08-19:
Hola Gabriel,

It should be possible just fine. I have that setup as many other people. I have multiple cameras hooked up to the same RPi3 and they are exposed to the internet each one using its own port.

Saludos,
Gaston