gdombiak/OctoPod #361: Siri suggestions should not include canceling a print

jessearmstrong, 2020-08-15:
I managed to cancel a print job today about 13 hours in because an OctoPod notification popped up (I don't even know what it was for, probably 50% announcement) and when i went to click on it there was a siri suggestion on the screen to instantly cancel my print. Which I tapped on by accident and instantly canceled my print. No confirmation warning or anything. I think canceling a print should be a more restricted thing that always require confirmation. I don't siri should be able to do it. I would hate for siri to activate accidentally and misunderstand me to cancel a print either. I don't know that I could reproduce it, but it was not fun.

gdombiak, 2020-08-15:
Ouch! That sucks! Sorry that happened @jessearmstrong. Since this action is done via Siri, I'm not sure I can add a confirmation prompt to confirm the action. I will look into that option and if not then see of not donating that action. Maybe via a setup configuration.

gdombiak, 2020-08-16:
[image]

gdombiak, 2020-08-16:
I'm not sure if change in configuration applies to already existing shortcuts. Might need to delete shortcut and create it again. Or maybe delete the app and recreate shortcuts. In my test I delete the app and things worked. Did not test before deleting the app. :(

commit 2e17dffd (2020-08-16): Intents/shortcuts for cancelling job or disconnecting from Palette 2 will ask user for confirmation before executing to prevent unintended actions.

Closes #361
closed 2020-08-16 by gdombiak