gdombiak/OctoPod #363: ngrok support

hhammarstrand, 2020-08-17:
I’m trying to connect with ngrok and the webcam stream is working fine, however I get some connection error and I can’t see print percentage etc. 

Is there a known solution to this? 

 
[image]

gdombiak, 2020-08-17:
Hey @hhammarstrand, is there a special reason for using ngrok over let's say ngnix? Are you looking into inspecting network traffic (AFAIK, it cannot inspect websocket traffic)?  

My recommendation would be to use nginx (with authentication) or [use VPN](https://github.com/gdombiak/OctoPod/wiki/Using-OpenVPN-for-Remote-Access). I haven't tested setup with ngrok or I'm aware of anyone using ngrok with OctoPrint. Having said that, OctoPod will use your URL to your Printer (OctoPrint) that you setup under Settings -> Printers and combine that to form the websocket URL. 

For instance, if you setup something like "https://myrandomurl.ngrok.com" in your printer then OctoPod will try to open a websocket connection to https://myrandomurl.ngrok.com/sockjs/websocket. You can test this from your browser using the developer console to see if a websocket connection via ngrok is correctly configured. I never used websockets with ngrok but from what I read, it requires you to use TCP (not HTTP) and  a proper authentication token. 

Hope this helps,
Gaston

hhammarstrand, 2020-08-18:
Hi @gdombiak 
The reason I'm using ngrok is the simplicity. There is a plugin for octoprint ([https://plugins.octoprint.org/plugins/ngrok/](https://plugins.octoprint.org/plugins/ngrok/)) and ngrok that automaticly creates a tunnel for me without having to do any configuration. 

I tried your suggestion and got the message `GET https://mynotsorandomurl.ngrok.io/sockjs/websocket 400 (Bad Request)`
The HTML pages says `Can "Upgrade" only to "WebSocket"`

I guess this means it won't work?

gdombiak, 2020-08-19:
Hey @hhammarstrand,

GET https://mynotsorandomurl.ngrok.io/sockjs/websocket will not open a websocket connection so that error is expected. To manually test from your browser you will need to open the developer console and execute something like this

> let socket = new WebSocket("wss://mynotsorandomurl.ngrok.io/sockjs/websocket");
socket.onmessage = function (event) {
  console.log(event.data);
}

hhammarstrand, 2020-08-19:
@gdombiak I see! You learn something every day. 

I tried the following: 

`let socket = new WebSocket("wss://url.eu.ngrok.io/sockjs/websocket");
socket.onmessage = function (event) {
console.log(event.data);
}`

And got this back:

`VM319:1 WebSocket connection to 'wss://url.eu.ngrok.io/sockjs/websocket' failed: HTTP Authentication failed; no valid credentials available
(anonymous) @ VM319:1`

gdombiak, 2020-10-05:
Hi @hhammarstrand, I'll close this ticket for now. We can reopen it if there is something to fix on the OctoPod side.

Thanks,
Gaston

closed 2020-10-05 by gdombiak