gdombiak/OctoPod #365: Use time/percentage from the PrintTimeGenius plugin

mneumark, 2020-08-19:
I have the PrintTimeGenius plugin. Among other things, it takes advantage of M73 on Prusa printers to give really accurate time/percentage estimates. It works great on the site but then I switch to Octopod and the time is out of sync (same with the percentage) and off by a significant margin. It would be nice if we could use that data instead in Octopod if available.

gdombiak, 2020-08-19:
Hey @mneumark,

I do not use that plugin but **M73 ETA Override** that does the same thing based of M73 present in gcode. I looked at the source code of the PrintTimeGenius plugin and things should work fine. Both plugins replace the out-of-the-box code of OctoPrint and use their own code to return remaining time. OctoPrint then exposes this information through its API (e.g. https://docs.octoprint.org/en/master/api/push.html#current-and-history-payload).

The fact that you also see the percentage of progress being off, I wonder if you are running into another error. Is the information not being updated? Do you see any error message in OctoPod about not refreshing?

Regards,
Gaston

mneumark, 2020-08-19:
I haven't seen any error, and I notice it moving....I'll try to take some screenshots so I can show you what I mean, maybe it'll help....

mneumark, 2020-08-19:
So you might be right about the time, it seems like they are in sync as far as I can tell. But the percentages are still a bit different, see this screenshots I took at exactly the same time:
[image]
[image]

Linuus, 2020-10-05:
Maybe it's the same thing as I see. Octoprint is showing 82% done and Octopod shows 52.5% done. The time is the same though.

gdombiak, 2020-10-10:
@Linuus that's very interesting. OctoPod does not calculate this information. It displays whatever OctoPrint is sending via websockets. If you see things getting refreshed in the app but still out of sync, could you open a ticket in [here](https://github.com/OctoPrint/OctoPrint/issues) and share this information? 

Thanks,
Gaston

Linuus, 2020-10-11:
> @Linuus that's very interesting. OctoPod does not calculate this information. It displays whatever OctoPrint is sending via websockets. If you see things getting refreshed in the app but still out [...]

Sure. I’ll report there if it happens again.

Just as an FYI it got to 100% at the same time as OctoPrint. Just that OctoPod’s percentage increased a lot faster in the end :)

tverett, 2021-02-03:
Not sure if this needs a new issue or not. I’m also using the print time genius plugin in addition to the time progress plugin that makes the progress bar time based instead of layer/file based. That being said the times and progress bars match in both the web ui and octopod app. However my notifications from octopod at 25%-50%-75% are all way off seemingly based on the old calculations based on file/layer.

HolisticDeveloper, 2022-10-18:
I am seeing a similar issue. Looking at what OctoPrint is sending over the websocket, I have these (truncated) messages:

```json
[
  {
    "current": {
      "progress": {
        "completion": 8.47088678070964,
        "filepos": 352370,
        "printTime": 6288,
        "printTimeLeft": 12048,
        "printTimeLeftOrigin": "genius"
      }
    }
  }
]
```

```json
[
  {
    "plugin": {
      "plugin": "DisplayLayerProgress-websocket-payload",
      "data": {
        "progress": "33",
        "m73progress": "35"
      }
    }
  }
]
```

So in this case, OctoPrint was showing 33% done whereas OctoPod was showing 8.4%. I'm assuming this will be an OctoPrint issue?

gdombiak, 2022-10-18:
Hi @HolisticDeveloper ,

You are correct. OctoPrint is the one calculating ETA and provides this information to OctoPod for rendering. Incorrect information should be reported to plugin developer of OctoPrint or OctoPrint dev if this is a core issue.

HTH,
Gaston

HolisticDeveloper, 2022-10-18:
I found a related feature request in OctoPrint (https://github.com/OctoPrint/OctoPrint/issues/4663). Looks like the main progress indicator is used for the gcode viewer.