gdombiak/OctoPod #366: I get 2 notifications for everything. How do I change that?

darrenph1, 2020-09-06:
For some reason I started getting 2 Notifications for everything. Is there a way to stop this?
[image]

gdombiak, 2020-09-07:
Hi @darrenph1,

My guess is that somehow you ended up with duplicated entries in your plugin configuration. Do the following to confirm if this is your case:
1. ssh into your octopi box (I'm assuming you are using octopi)
1. cd .octoprint
1. Edit config.yaml and go to plugins: -> octopod -> tokens
1. You should see an entry for each combination of printer and iOS device. If you have duplicates you will see more entries than you expected
1. Feel free to delete all tokens and restart both OctoPrint and OctoPod app. OctoPod will register with new tokens again

People that tried to spawn a new OctoPrint instance by doing a backup and restore of the configuration are the ones that ended up duplicating entries. Not sure if that was your case. There is a ticket to show in OctoPrint UI the list of registered tokens so you can manage them from there.

Hope this helps,
Gaston

darrenph1, 2020-09-08:
>    2. cd .octoprint
>    3. Edit config.yaml and go to plugins: -> octopod -> tokens
>    4. You should see an entry for each combination of printer and iOS
>    5. Feel free to delete all tokens and restart both OctoPrint and

Thanks, I could never find where the location is via the GUI but via ssh I
found the tokens and deleted them. I'll know in about 6 hours if it worked
but sound like it should. I noticed the " - " that you mentioned also in
other plugins I have. I guess that means I have duplicates for those as
well.  I should be able to delete those also, yes?

Thanks for your help and the great app!!

Darren

gdombiak, 2020-09-09:
@darrenph1 , remember that a restart of OctoPrint and also of the OctoPod app are going to be needed for the changes to make effect and have the phone registered to receive notifications from the plugin.

The "-" that you see in that file means that the property (e.g. tokens) has many values. Under "plugins" you also see that each entry has a "-" which means a value of the list of plugins. In [yaml this is called a collection](https://www.tutorialspoint.com/yaml/yaml_collections_and_structures.htm).

darrenph1, 2020-09-09:
That fixed the issue. Thanks!

closed 2020-09-09 by darrenph1

spartan152, 2024-12-24:
I'd like to re-open this as I was not able to resolve this. I only see one token in my config.yaml file, although I have restored from a backup a couple times. Should I try to delete the token and set it up fresh?

gdombiak, 2024-12-25:
Hi @spartan152, the instructions on this ticket are no longer necessary. You can now use your browser to go to OctoPrint's settings. Go to the OctoPod Notifications page and click on Devices. That page shows all the devices that are registered with that OctoPrint instance. You can click on the trash icon to delete any device you no longer want to be notified. If you delete all the entries then open the OctoPod iOS app and it will register again.

Hope these instructions help.
Gaston