gdombiak/OctoPod #368: Keep display awake when webcam is in fullscreen

vordenken, 2020-09-12:
Hey,

first of all I really love this app. 
One thing that annoys me a little bit is the fact that the display doesn't stay on when I fullscreen the webcam. Is this something you would add as a feature?

gdombiak, 2020-09-12:
Hi @vordenken ,

Under Settings --> Appearance you can disable _Auto-Lock when idle_ and the app will not let the iOS turn off the screen when the app is used. It is not the exact same thing you asked but the closest.

Would this solve your need?

Regards,
Gaston

vordenken, 2020-09-13:
Hi @gdombiak ,

thanks for your answer. I know that this is a feature in iOS but this would mean my phone wouldn't auto-lock at all. 
What I mean is that the display only been kept on while displaying the webcam in OctoPod. Similar to when watching Youtube, Netflix or any other video-type content.

gdombiak, 2020-09-14:
hey @vordenken ,

I'm referring to a setting inside of OctoPod itself. Go to Settings --> Appearance in OctoPod app to control this setting. Let me know if this would help with your request. I still see value in your request since that could help others with the same request.

Thanks,
Gaston

commit 6cafdaa9 (2020-10-09): Do not let screen turn off when app is idle and camera is in full screen

Closes #368
closed 2020-10-09 by gdombiak