gdombiak/OctoPod #369: Option to rotate webcam

vordenken, 2020-09-12:
I use octoprint with a webcam that has portrait format instead of vertical format, so it looks really weird when I toggle fullscreen. Heres a screenshot: 
[image]

Would it be possible to add the option to rotate the webcam/video so it can fit the screen?

gdombiak, 2020-09-12:
Hey @vordenken,

I wonder if going to OctoPrint's web UI under Settings --> Webcam & Timelapse and check _Rotate webcam 90 degrees_ will help solve this. Not sure but worth a try. OctoPod follows the configuration that you have for the webcam in OctoPrint.

Thanks,
Gaston

vordenken, 2020-09-13:
Hi @gdombiak ,

I already use this feature to make my webcam show up in the right direction within octoprint. Here's what it looks like:
[image]

I think OctoPod also displays it "correct" in that matter, but because my webcam is in portrait mode, this big grey bars appear on the side. It would be awesome to rotate the webcam within OctoPod only, so it uses the whole screen.

novski, 2023-03-30:
I agree, my cam view acts exactly the same way.
If i tap it to enlarge it turns an gets letterboxes over 60% of the screen.
It should not rotate if the cam is set to rotate in octoprint.
[image]

[image]

[image]

This happens when i tap the videobox:

[image]