gdombiak/OctoPod #371: Placeholder image in widget

sdaitzman, 2020-09-19:
[image]

Is this expected behavior? I expected it to show a video here while printing, but it doesn't. The webcam feed appears as usual in the app itself.

gdombiak, 2020-09-19:
HI Sam,

I created [this video](https://youtu.be/QkWxQr8Glqo) that shows how to configure the widget to use any of your printers or cameras. I guess that many others will have the same question so I also updated the [FAQ](https://github.com/gdombiak/OctoPod/wiki/FAQ-using-the-app#how-do-i-use-use-the-new-ios-14-widgets).

I tried to auto-select the first camera to avoid this confusion but I couldn't figure it out. Not sure if it is me or the new widgets API is still not mature enough. Probably me. :)

sdaitzman, 2020-09-19:
Amazing, thank you so much! It worked! 😄

closed 2020-09-19 by sdaitzman

sysadmininator, 2020-09-20:
This worked for me as well. Thanks!

I want to say it selected my first printer (my only printer... for now ;)) but didn’t select my only/default camera. It wasn’t until I searched for this issue that I noticed the camera option as well. There’s a chance I tried to configure the widget while my octoprint server was offline so perhaps the cached config of the printer selected fine but camera was a no-go due to it not being available when octopod was trying to make a decision. 🤷‍♂️

jollino, 2020-10-11:
I was bitten by this too, and only randomly discovered that there are widget settings (not Octopod's fault, it's just not obvious in iOS 14).
However, in my humblest opinion, just like the widget selects the first/only printer, it should also select the first/only camera... especially if there's just one of each. :)

gdombiak, 2020-10-11:
Hey Daniele,

I fully agreed. I tried so many times and different ways to figure this out and still no luck. Tried in latest iOS version and still not working as I assumed. 

Swift should be executing ```#defaultCamera(for intent: WidgetConfigurationIntent) -> WidgetCamera?``` and pass as a parameter the default/selected printer and I return a default camera. The 2 problems I see is that I do not receive a selected camera and whatever I return is not displayed by the widget. The method is called but I do not see the result being used.

I will keep looking and reading into this but so far no luck. I'm sure many people do not figure out they need to edit the widget to set the video to display. :(

jollino, 2020-10-11:
Does this also happen in 14.2 beta? With all the issues that 14.0 had, I wouldn't be surprised if it's just a glitch in the API.

As an alternative, would it be possible to show a brief explanation inside the widget itself when it's first loaded, so that people know how to look for the options? Sort of a "modal layer" with text that you dismiss, if that makes sense.

gdombiak, 2020-10-11:
I like that idea @jollino. Trying to minimize ask to translators .... what do you think if I overlay the "Configured Widget" text over the place holder image when no camera was selected? Maybe I can add it below the image so it is easier to read?

jollino, 2020-10-11:
That would work! Maybe something about reading the FAQ too, because Apple made it really confusing. I've been extremely busy lately so I hadn't fiddled with the widget much, and only accidentally discovered the widget options screen when I was setting up the iPad I just got!

gdombiak, 2020-10-11:
Will add the text since that will orient people on what to do. Widgets are super limited. :( They are somewhat like complications in Apple Watch. They run once to display content and if you tap on the widget it opens the app. This means that we cannot stream video even for a few seconds or add logic to run on gestures. :(

gdombiak, 2020-10-11:
[image]