gdombiak/OctoPod #372: Flow and federate sliders at 75% when starting app

perfessor101, 2020-09-20:
Hello,
I’ve got Flow and federate sliders at 75% when starting app and the printer seems to be under extruding.  
I slide both to 100% and my print starts coming out better

Not sure if it’s actually sending the federate and flow before I touch it. 

When cancelling print it doesn’t reset the extruder ... if I start a print without sending the gcode for extruder reset it rewinds the complete distance printed before starting to extruder.

gdombiak, 2020-09-20:
Hi @perfessor101,

OctoPrint cannot read current flow/feed rate from the printer since Marlin does not expose this information. The value that you see in OctoPod is not the current flow/feed rate of the printer but the last value that you set using the app. By default it will be at 100% in the application. If you move the slider or set a new value then OctoPod will ask OctoPrint to set that value to the printer.

In short, OctoPod only sends a new flow/feed rate when the user moved the slider or entered a value.

Hope this info helps.
Gaston

perfessor101, 2020-09-20:
When I close and restart yesterday’s updated version of octopod reads 50% feedrate and 75% flow

If it doesn’t send until changed all is good

Thanks for your time

gdombiak, 2020-09-20:
That's correct. It will only send it when user made a change. Otherwise it will show whatever you set last or 100% if you killed and restarted the app.

gdombiak, 2020-10-05:
Hi @perfessor101, I'll close this ticket unless there is something you want me to fix. We can reopen if that is the case.

Thanks,
Gaston

closed 2020-10-05 by gdombiak