gdombiak/OctoPod #373: Issue with Enclosure Outputs in Octopod iPhone App

OCOLLET91, 2020-09-21:
Hi,
I use Octopod + Enclosure to be able to switch on/off the light on my printer and to cut power in case things go wrong.

Up to now it works quite well with Safari but I report the following bug with the iPhone app of Octopod on "Enclosure" tab:

I open Octopod on my iPhone
Then I go to the Enclosure tab.
There I can switch On or off my leds and eventually the printer.
You see on pictures at 20:12 that there is no light but Octopod report that LED is ON on one picture and  OFF on the second picture. The difference between the 2 pictures has been to close and re open the App...

In fact the sequence was:
LED were ON in reality
Then open Octopod on iPhone
Then switch LED Off via iPhone interface
Then close octopod App on iPhone
Then re open octopod app on iPhone: Leds are still off in reality but reported ON on the APP... to switch them on b-via the app you have to click on "off" then confirm then click ON...!...

As far as I have tested, it seems that the issue appears only when the App is closed and reopen, the status becomes wrong.

And you see at 20:20 a picture when leds are on and reported on.

Thanks a lot for this very useful app. I am not a software guy so I can only report issues and not able to solve them...

Tell me if you need further information

Regards

Olivier

[image]
[image]
[image]

gdombiak, 2020-10-07:
Thanks @OCOLLET91 for the detailed report. I do not use this plugin so took me some time to simulate it in my environment. I was able to reproduce and I see what is going on. I will fix it for the next release.

Merci!
Gaston

closed 2020-10-07 by gdombiak

OCOLLET91, 2020-10-07:
Hi Gaston

Thank you for this amazing app!!!

Kind regards

Olivier

commit 13643226 (2020-10-07): Refresh Enclosure outputs when we receive a notification via web sockets stating that outputs have changed their state

Closes #373