gdombiak/OctoPod #383: Feature Request: iOS picture in Picture

Drakurius, 2020-10-05:
Hey there,

Not sure if possible but iOS on iPads and now even on iPhones allows some sort for Picture in Picture. For example if I watch netflix and go back to the homescreen the video is minimizing and still there so I can do something else while still watching. 

Is this something you could implement in octopod so I could watch the print while still doing something else?

That would be awesome!!

Cheers from Germany

gdombiak, 2020-10-05:
I love this idea @Drakurius. Started to look into it and it seems like Apple does not let developers control the video so need to use what Apple offers. That means that we are limited to what [AVPlayer supports](https://programmersought.com/article/75621732637/) which means MJPEG does not work. 

I will do some tests but it seems like we will need to install some extra services on the Raspberry Pi to convert MJPEG into a supported format. Problem is that not many people will do this step and we need to see how well can RPi handle the extra CPU and memory load.

Will keep exploring this since I think it has a lot of value. 

Thanks,
Gaston

Drakurius, 2020-10-07:
Thanks for looking into what is possible. I have no idea!!

gdombiak, 2020-10-07:
Me neither. The fun part of each request is that it takes me on a journey where you learn new things. :)

gdombiak, 2020-11-24:
Now that OctoPi and OctoPrint will support HLS feeds .... I can now implement this feature. :D

Drakurius, 2020-11-24:
Awesomeet me know when you need someone to test 👍

commit 770bd6de (2020-11-29): Clicking on Home button will no longer start PIP automatically.

Closes #383
closed 2020-11-29 by gdombiak