gdombiak/OctoPod #384: Siri suggestions not showing up in settings

tomavisar, 2020-10-09:
Hey, running iPadOS 14.0.1 and the Siri suggestion won’t appear in the settings. Tried deleting and re adding them using the in app buttons and reinstalling the app but nothing seems to work.

gdombiak, 2020-10-10:
Hi @tomavisar,

To add new Siri commands open Shortcuts app and click on + at the top right. Click on + Add Action button and search for OctoPod. It will list entries for which you can create a new Siri command. Select one and then click Next. Give it a name which is the phrase you will say to Siri. Once done click Done and you are good to go.

I just updated https://github.com/gdombiak/OctoPod/wiki/Siri-integration-with-OctoPod to explain how it works in iOS 13 and iOS 14. Let me know if things are still not working.

Thanks,
Gaston

gdombiak, 2020-10-25:
@tomavisar did this issue got resolved with the suggestion made above?

gdombiak, 2020-10-31:
@tomavisar will assume suggestion above solve this ask. We can reopen the issue if that’s not the case

closed 2020-10-31 by gdombiak