gdombiak/OctoPod #390: iPad iOS 14 Can’t add new printer on latest build

Jon-So, 2020-10-15:
Thank you Gaston for your dedication and hard work

gdombiak, 2020-10-15:
Thanks @Jon-So. Could you share more info about what is not working or how to reproduce this?

Regards,
Gaston

Jon-So, 2020-10-16:
I couldn’t add any new printers via iPad iOS 14. And the save button wouldn’t light up for me after I put in all the necessary info.

Jon-So, 2020-10-16:
Hey Gaston, I think I figured out what the problem was, probably something to do with my network, I changed the hostname and it allowed me to add it after I restarted the pi... thanks for all you do...

gdombiak, 2020-10-16:
Thanks for the update @Jon-So . Is there anything I need to fix or can we close this ticket?

Regards,
Gaston

Jon-So, 2020-10-16:
As far as I can tell it seems to be ok when I changed the hostname but I can’t understand why that would have been an issue... I think it comes down to having multiple printers... but I am ok with you closing it for now, if I run into anything I will let you know. Thanks for the amazing app man!

gdombiak, 2020-10-16:
Cool. If you find a way to reproduce the disabled button even after completing the 3 fields with valid data then we can reopen this ticket. I tested it on my ipad with latest os and worked fine. We might be missing some edge case that needs to be fixed

closed 2020-10-16 by gdombiak