gdombiak/OctoPod #394: Apple Watch : there is no camera

Eldoc62, 2020-10-23:
Hi,
I’m all update
Apple Watch 4 
In octopod on iOS it’s ok I’ve the picture of my camera but not on Apple Watch...
There is a circle green but no picture of my camera.
Thanks

gdombiak, 2020-10-24:
Hmm, I wonder if image size is too big and it fails to reach the Apple Watch. Could you try reducing resolution of the camera and see if things start to work? You can revert the change after this experiment. It will help collect info to understand what is going on.

Thanks,
Gaston

nikomaxi, 2020-10-28:
I have also the problem. I have reduced the resolution of the camera without success.. 
I have reinstalled all Apps (iOS and watchOS). I have reinstalled OctoPod extension on Octoprint...
without success. 
My widget does not load things automatically. I first have to open the App...
Camera is still not working!!

gdombiak, 2020-11-26:
While adding HLS support to Apple Watch, I found a few use cases that would not render any image on the Apple Watch and the refresh button remains disabled. I'm not sure if this is your use case or not. This would happen if Apple Watch is not able to connect to iOS device to fetch image or if there was an error while sending the image from the phone to the watch.

I fixed those 2 cases in the next release. Let me know if you want to be part of beta testing for the next release.

Thanks,
Gaston

commit ebfb5f0e (2020-11-26): Improved/fixed logic to fetch camera image. There is a retry in case of a transient error. In case of an iOS error and HLS then a new error message is displayed

Closes #394
closed 2020-11-26 by gdombiak