gdombiak/OctoPod #395: Bar graph

Jon-So, 2020-10-23:
Hey Brother I was printing with my ender 3 pro last night and I noticed the bar graph with the print percentage was completely different than the printers, is that normal? The print was at around 52% and then in octopod it was at 30% and then in the dashboard it was another number there too. Not really a big deal but was wondering where the percentage comes from?

gdombiak, 2020-10-24:
Hey @Jon-So, this has become a frequent comment so I'm starting to suspect something is going on. Maybe you can help me get to the bottom of it?

OctoPod gets the information from OctoPrint. OctoPrint pushes the notification via websockets to refresh the UI of OctoPod, the webUI or any other client. If OctoPod is behind then that means that the information is not being sent via websockets. Could you look at OctoPrint logs and check for any errors? Feel free to post them here and I can check them for you if you want. 

Do you know what changed recently? Did you install some new plugin? New version of OctoPrint? Some plugin got updated? Since when this started to happen?

Thanks,
Gaston

fergusontrevor, 2020-10-28:
My octoprint said it was approximately 2.5 hours. My ender 5 said it was 2 hour 37 minutes. This app says it is almost 5 hours.

gdombiak, 2020-10-29:
Hey guys,

I've been doing some more testing and I was able to see a number in OctoPrint and a different _print time left_ in OctoPod. However, here is something interesting that worth checking in your case. If you look at the picture below, you will see that OctoPrint displays a value but if you hover over the _print time left_ you will see another value. OctoPrint is sending to its clients, like OctoPod, the value that you see when you hover.

Could you check if this is your case as well?

[image]