gdombiak/OctoPod #399: Layer Notifications

amodernstitch, 2020-10-25:
Is there a way to be notified when printing begins? I tried layer 1 notification but that notifies once layer 1 is complete. 

Thanks so much.

gdombiak, 2020-10-25:
Hi @amodernstitch,

At which precise moment are you referring to "print begins"? 
1. Is it when you clicked Print? -> This is possible
1. Is it after it heated and started to extrude? -> I do not think this is possible unless you analyze gcode
1. Any other moment?

Thanks,
Gaston

amodernstitch, 2020-10-25:
its number 2... after it is heated and started to extrude. 
 
I think it could be achieved with setting a notification when temp rises to a certain point like the notification when the bed reaches a certain temp. Would that be possible?

gdombiak, 2020-10-25:
Gotcha. I think that https://github.com/gdombiak/OctoPrint-OctoPod/issues/54 is exactly what you are asking for. It is scheduled for the next release of the plugin. Would that solve this ask?

Gaston

amodernstitch, 2020-10-25:
That would do it!

closed 2020-10-25 by amodernstitch