gdombiak/OctoPod #405: Issue controlling Enclosure Plugin with Octopod

OCOLLET91, 2020-11-12:
Hello

I have the following configuration:
Octopring with Enclosure plugin to control my Prusa MK3S printer and with relays on GPIO to control LED lights and Printer supply
When I control from Octoprint dashboard, I control flawlessly the 2 relays on/off

Now I installes Octopod on my iPhone to monitor my printer, switch on/off the light when I am away from home.
It works perfectly. I even see the temperatures sent by Enclosure plugin. 
BUT:
I just have an issue with the control of the 2 relays controlled bu Enclosure on GPIO ports. At first look it seems erratic. What I finally found is that the indication of the status ON/OFF of the relays is erratic: on the 2 attached pictures you see the status of printer is the same: LEDs are OFF (all black) and the printer is ON (Blue light at bottom right)
Now if you look carefully those 2 pictures you see that on on e picture both LED and printer are indicated OFF (red cross) and on the other one they are reported ON (No red cross...)
Then if I want to switch ON I have to play with the interface until the red cross is on the LED  then switch ON...

Something is not working correctly... can you help or advise? Probably I do something wrong?

This is the only issue I have with this wonderfull app

Thanks for your help!
[image]
[image]

Note: IOS 14, All softwares plugin and apps at the last known issue.

gdombiak, 2020-11-12:
Bonjour @OCOLLET91,

I'm not sure I have enough information to know how to reproduce what you are seeing. Could you describe in detail the steps that I should follow to reproduce this?

Merci,
Gaston

OCOLLET91, 2020-11-13:
Bonjour Gaston

I will try. It is not easy since I find it erratic the behavior is not
always the same.

Let’s say for exemple that my printer is on, printing a part and I want to
monitor with the cam.

I open octopod, go to the page with the enclosure switches and for example
I see that both switches are on... but led is off one printer is on as
below...:
Then as enclosure and or octopod believe the LEDs are on it is not possible
to switch on the LEDs...

Then I have to play several times (1,3,5,10 it depends...?...) with the led
switch to reach a state were the LEDs are shown off with the Red Cross. It
often happens that the printer stows off at the same time when it is in
fact on as you can see in picture below:
(Blue light in the corner shows the printer on)

Then I can press the LEDs switch and finally have the light on in reality:

Very strange behavior... only with octopod... with Octoprint web page I can
control Enclosure relays without any issue...

Is it more clear for you?

Kind regards

Olivier

spatcho, 2020-11-15:
Hi,

I have also the issue, it's like when you turn on or off an enclosure io, printer and light in my case, the app doesn't refresh / reflect the new status immediately, even after restarting the app. And so you can't turn on (or off) again till the app is refreshed. Using octoprint directly via web interface it work without problem. No time yet to build and test from Xcode.

Best

Bernard

spatcho, 2020-11-15:
> Hi,
> 
> I have also the issue, it's like when you turn on or off an enclosure io, printer and light in my case, the app doesn't refresh / reflect the new status immediately, even after restarting the app. And so you can't turn on (or off) again till the app is refreshed. No clue why and when it refresh finally, but it does. Using octoprint directly via web interface it work without problem. No time yet to build and test from Xcode.
> 
> Best
> 
> Bernard

Ok, it seems after a rest query for the value of one io, both entries where filled with the result.

I did modify EnclosureViewController.swift like that and it work for me :

    fileprivate func refreshOutputsStatus(done: (() -> Void)?) {
        // Fetch current status for GPIO outputs
        for output in self.outputs {
            **let currentIndex = output.index**
            if output.type == "regular" {
                self.octoprintClient.getEnclosureGPIOStatus(index_id: output.index) { (value: Bool?, error: Error?, response: HTTPURLResponse) in
                    if let value = value {
                        for (index, output) in self.outputs.enumerated() {

                            **if (index+1) == currentIndex** {
                                // Update status in wrapper
                                output.status = value
                                // Refresh only row of updated status

                                NSLog("refreshOutputsStatus : \(output.label) is \(output.status)")
                                
                                DispatchQueue.main.async {
                                    self.tableView.reloadRows(at: [IndexPath(row: index, section: 0)], with: .automatic)
                                }
                            }
                        }
                    }
                }
            }
        }
        // Execute done block when done
        done?()
    }

OCOLLET91, 2020-11-15:
Hello Bernard

I understand from message below that you solved the issue?

How and what shall I do in clear? I am not a software man so your message
is a bit as Chinese for me...

Thank you in advance

Kind regards

Olivier

spatcho, 2020-11-15:
Hi Olivier, Let wait until Gaston review it, Best Bernard

gdombiak, 2020-11-15:
Thanks guys for the extra info and having a fix. I’m now running. Will check it out later and submit a new release soon with the fix. 

Gaston

OCOLLET91, 2020-11-15:
Thanks a lot Gaston!

Keep us informed when release the update. Not an emergency at all, just a
little problem with a great app!

Thanks for creating it

Kind regards

Olivier

closed 2020-11-16 by gdombiak

OCOLLET91, 2020-11-16:
> Closed #405 <https://github.com/gdombiak/OctoPod/issues/405> via 963ea19

Hi Gaston

Does it means you released a new version?

Where do I find it?

Thanks

Olivier

gdombiak, 2020-11-16:
New version is now in the App Store. Let me know if that fixed this issue. 

Merci
Gaston

OCOLLET91, 2020-11-16:
Hi Gaston,

It works perfectly now!
Thanks a lot, you are a Magician for me!!!

Kind regards

Olivier Collet

gdombiak, 2020-11-16:
Great to hear! Thank @spatcho for quickly finding the problematic area and submitting a fix!

If you haven't already please leave a review in the app store and share with friends! I make no profit from this project and would like to keep helping as many people as possible. :D 

Merci,
Gaston

OCOLLET91, 2020-11-16:
Done ****_ ratings and a good comment in the app store!

Thanks a lot for this wonderfull App Gaston!

commit 963ea19b (2020-11-16): Enclosure output is now correctly updated. Only the target element is now updated. Thanks Bernard Spatcho

Closes #405