gdombiak/OctoPod #407: DNS Query Pulls from Public DNS

aethernex, 2020-11-16:
This seems to be a recent issue, but I have split horizon DNS configured on my internal network so that I can use a single FQDN for my printers. All DNS lookups work fine at an OS level on all my device, except iOS devices. If I kill all my apps and flush DNS on my iPhone, I can then perform a lookup and it reports the proper internal IP address. I can use any number of other apps without a problem, but the moment I launch Octopod my iPhone's DNS table changes my FQDN resolution to my public IP instead of the internal IP address. This then breaks communication and sends traffic outside my network for no apparent reason.

Steps to test:

Kill all apps
Flush DNS on iOS device
Ping FQDN
Receive internal IP address
Launch any app that connects to local IPs with success
Ping FQDN and see appropriate IP
Launch Octopod (fails)
Ping FQDN and see public IP

Current workarounds:

Use cellular data to connect
Enter IP address in configuration

This isn't ideal, and oddly it works fine on my wife's printer which is an identical hardware and software configuration. It does happen to multiple iOS devices, so it's not my particular iOS device. And I'm sure the network is fine as every other device receives and keeps the appropriate DNS resolution. 

Thoughts?

gdombiak, 2020-11-24:
Hi @aethernex,

Thanks for sharing a detailed report. Unfortunately, it is hard to tell how to achieve your desired effect. OctoPod does not **directly** uses DNS or changes which DNS server to use or flushes or populates DNS. I use the [standard network SDK](https://developer.apple.com/documentation/foundation/url_loading_system) for making HTTP requests. Apple's libraries do the hard low level work. Maybe check in Apple dev forums?

Sorry for such an unhelpful answer. :(

Gaston

aethernex, 2020-11-24:
Thanks for the reply. I honestly expected this to be the answer, so don't
worry about it. I figured you used a standard implementation, it's just so
weird that it happens nowhere else, even other Octoprint installations. I'm
going to try rebuilding the OctoPi on a Pi4 and configure my reverse proxy
to handle traffic instead of a direct connection to the Pi. Then I'll try a
clean install of iOS in case there's something messed up in my devices if
that doesn't work. Then I'll reach out to Apple if none of that works.

Take care,

gdombiak, 2020-11-24:
Sounds like a solid plan to me! Good luck and keep me posted!

Gaston

closed 2020-11-24 by gdombiak