gdombiak/OctoPod #408: Very high data usage over cellular

BlaydeRunner08, 2020-11-22:
I recently noticed that I'm noticing that OctoPod seems to take an incredible amount of cellular data.

Typically, I used OctoPod to keep an eye on my print while I'm on the same wifi, but occasionally I use it to monitor it remotely.  Usually, when I'm out, I'll open the app, check the print/webcam, and then turn my phone off.  However, I just went to check my data usage, and I noticed that OctoPod used up 2GB in the last month!!!

Does the app continue to run the webcam stream even when it's running in the background?  Has anyone else experienced this? 

[image]

gdombiak, 2020-11-22:
Possibly good news. While adding HLS support to OctoPod I noticed that (at least in the Simulator) the websocket connection remains active when the app goes to background. iOS should have stoped network traffic but it did not at least when using the simulator.

I added code so that the app closes the websocket connection when it goes to background to make 100% sure that there is no more traffic.

BlaydeRunner08, 2020-11-22:
That’s great to hear! Thanks for all your work!

closed 2020-11-22 by gdombiak

gdombiak, 2020-11-26:
@BlaydeRunner08, are you using the Apple Watch app? Are you using OctoPod with multiple printers?

I fixed a few more cases where there is network traffic when changing selected printer from Apple Watch. And even more if you have a camera configured. I now see zero network traffic after the last fixes

BlaydeRunner08, 2020-11-26:
@gdombiak Yes to the Apple Watch, but no to the multiple printers.

gdombiak, 2021-01-17:
@BlaydeRunner08, I'm curious to hear if the new release actually fixed the high data usage you observed over cellular.

Thanks,
Gaston

BlaydeRunner08, 2021-01-17:
Hey @gdombiak,

yeah, it looks muuuuuch better. These are my cellular stats, last reset around the time the new release:
[image]

Thanks again!

gdombiak, 2021-01-17:
Sweet! That's great to hear. Good to confirm our wild guess hit it on the head. :)

Thanks,
Gaston

commit 6ad54623 (2020-11-22): Disconnect web socket when app goes to background

Closes #408