gdombiak/OctoPod #410: I can’t add my printer

renegadecitizen, 2020-11-23:
I’m trying to add my printer
I’m using octoprintanywhere
I scan the code
But it gives me the connection error 
The operation couldn’t be completed 
(Starscream.WSError error 1)

Could you please help me?

gdombiak, 2020-11-24:
Hi @renegadecitizen,

Are you using https or http to connect to OctoPrint? If you are using https and you are not using a valid certificate ... then this error means that the presented certificate is not valid. If you want to use https with not a valid certificate then go to OctoPod -> Settings -> Security and disable certificate validation.

Let me know if that helped or if you are not using https.

renegadecitizen, 2020-11-24:
I tried it
It didn’t help 
It gives me a warning 
It couldn’t connect because it’s not in the correct format 

Thanks for helping man

gdombiak, 2020-11-24:
Could you share the warning that you got and the incorrect format message that you got? That might help troubleshoot this issue.

Thanks,
Gaston

renegadecitizen, 2020-11-24:
[image]
[image]

These are the two errors I’m getting 
It’s not a local installation 
My Pi is located at
https://renegadecitizen.octoeverywhere.com

gdombiak, 2020-11-25:
How does thing work when you are in the local network? If they work fine, then we should take a look at your reverse proxy (I'm assuming you are using one). Are you using a VPN or port forwarding? Could you share more about your setup.

Regards,
Gaston

renegadecitizen, 2020-11-26:
It works fine on local
Neither VPN or port-fowaring
Only octoeverywhere.com
I understand if it doesn’t work with your app

gdombiak, 2020-11-26:
Gotcha. I checked out that product and he developer is slowly adding support for apps. Check out [this link](https://www.reddit.com/r/OctoEverywhere/comments/jy08e5/octopod_support/). I think he also mentions that on his website

renegadecitizen, 2020-11-26:
Cool they mention octopod so there’s hope

gdombiak, 2020-11-26:
Will close this ticket for now. Let me know if you need any further help.

Gaston

closed 2020-11-26 by gdombiak