gdombiak/OctoPod #411: iOS app not available on AppleSilicon Macs?

Technofrikus, 2020-11-24:
Hello,

I absolutely love the app and use it on my iPhone all the time. But now with the AppleSilicon Macs it would be great to also run it on them. Would be great for monitoring in the corner of the screen and having one less tab open. And probably less power draw then the browser.

At the moment it seems it was opted out of the AppStore. Why is that? Is it planned to add it later?

Thanks!

gdombiak, 2020-11-24:
Hi @Technofrikus, thanks for bringing this up. I just configured the App Store to let you run OctoPod app in AppleSilicon Macs. Those M1 look very attractive .... if only they had 32GB and more ports. I'll have to wait for MBP 16'' hoping to have those features

closed 2020-11-24 by gdombiak

Technofrikus, 2020-11-24:
Wow, that was fast, thanks! Works great :) Even the window is resizable. That doesn't work with several other iOS-Apps. 

RAM doesn't seem to be such a big issue. Performance is not impacted as much as I would have thought. But more ist still better of course.
https://www.youtube.com/watch?v=h487I_5xOZU

gdombiak, 2020-11-26:
Nice to hear you got it working. It would be great if you could record it running on the new AppleSilicon Macs. Would you be ok if I share your video (if you make one)?

Thanks,
Gaston

Technofrikus, 2020-11-26:
Sure can do, no problem :) What do you want me to show? Just clicking though the App and show everything once? Or something else?

Technofrikus, 2020-12-04:
Here, I made something :) Hope it helps and is what you were looking for.
https://wenecloud.wenedeux.de/index.php/s/pKR7B6FronRfD3P

gdombiak, 2020-12-04:
That video is very good. I was thinking of doing the same exact video but showing that it is running in your Mac. Maybe show the desktop and the app running so people can see that it is not a recording on my phone but on a Mac? 

If you are ok with that video, are you ok if I share it in Twitter and Facebook groups? I think this is very impressive!!! More people will want the M1 now. ;)

Technofrikus, 2020-12-05:
Now with the complete display visible and some resizing action. Feel free to use it anyway you want. I'm glad to be able to help :) 

https://wenecloud.wenedeux.de/index.php/s/qdNBr5RLPCxAgFy

gdombiak, 2020-12-06:
That video is AWESOME!!!! Thanks so much @Technofrikus for sharing. Have you joined the new [OctoPod Facebook group](https://www.facebook.com/groups/octopodprint)? I will share your video there, on youtube and Twitter. I will mention you as the author.

Technofrikus, 2020-12-06:
Glad you like it :) I just joined the group. Happy promoting :)