gdombiak/OctoPod #414: No camera feed on iPad Air 2020

pixelmagic66, 2020-11-27:
On my older iPad Air 2 I get a camera stream above the printer status on the first page, but aomehow with the exact same settings on the Air 2020 no camera feed, just the printer status. Even side by side checked the settings, deleted from Icloud and new config the same result.

Any clues on how to fix it ?

gdombiak, 2020-12-01:
That is indeed very weird. Hardware should be indifferent to the app. I will test it on the simulator (I do not have that hardware with me) and will report back.

One thing to check but I'm 100% sure it is not related: Go to iOS Settings -> OctoPod and check that settings are the same. Maybe for some reason Local Network is not enabled in the Air 2020?

ssnyder30, 2020-12-05:
This also doesn’t work for me in landscape orientation on an iPad Air 2020. However it does work in portrait orientation. Rotate the iPad one way, camera feed is visible, rotate it the other way, and the camera feed disappears.  I suspect this has to do with the sizing of UI elements to stretch to fill the screen on rotation. Perhaps the Air 2020’s resolution is involved?

gdombiak, 2020-12-05:
This info helps a lot to troubleshoot this. Will add it to the next release since I might have an idea on what is going on. Thanks @ssnyder30 for the extra info.

pixelmagic66, 2020-12-05:
I can confirm that it works in portrait mode and not in landscape on the Air 2020

commit cfdb523f (2020-12-05): Set dimensions for camera feed for new 2020 hardware.

Closes #414
closed 2020-12-05 by gdombiak