gdombiak/OctoPod #416: OctoPod does not connect anymore / add new printer not working

Uwe-Kaki, 2020-11-30:
Hi, after some time I used OctPod again which was both on iPhone & iPad working perfectly.
Now it is not connecting to my RasPi Octoprint any more : reinstalled the APP already, deleted the printer to add again.. all not working.
Both iPhone and iPad newest IOS, OctoPi / Octprint newest release.
"Alternative" IOS Apps I tested work straight away.

Any idea what happened or what I can do to fix it ?

many thx in advance

gdombiak, 2020-11-30:
Hi @Uwe-Kaki,

By any chance, does the hostname end with / ? If it does, try removing the last / and let me know if that fixed the issue. If you were not using a / at the end, could you share the error that you see when failing to connect? Maybe a screenshot?

Thanks,
Gaston

Uwe-Kaki, 2020-11-30:
Hi Gaston

Thx for the quick response

 

I have deleted my ( one and only ) printer out of the settings , as I expected some glitch due to new versions both of OctoPod& Octoprint and was hoping I could simply start from scratch -> no errors to show, sorry

 

But here some screenshots of what I did and see.

 

A few weeks ago everything was working as expected, even with individual API Keys from OctPrint Appliction on the PI.

I named the attached JPGs to explain what it is about. 

I tried every hostname with / without http, slash, only ip adresse etc .. I always get this strange „notepad“-like popup.

There cannot be a general issue with my network, Octoprint setup etc as an alternative IOS App which I downloaded for the time beeing worked immediatly 

 

 

LG aus dem Norden

Uwe Knöfel

gdombiak, 2020-11-30:
Hi Uwe,

I think that images are not attached when replying via email. Could you reply directly in github and attach the screenshots?

Vielen Dank,
Gaston

Uwe-Kaki, 2020-11-30:
Hi Gaston, here the pictures.
[image]
[image]
[image]
[image]
[image]

gdombiak, 2020-11-30:
I'm a little confused. Is the issue that the "Suchen" does not show anything? Or is the issue that after completing the first 3 fields in the form (name, hostname and API Key) things do not work when you go to the main panel?

"Suchen" (Scan) is a convenient way for finding a printer in the local network and autocompleting the name and hostname (the first 2 form fields). You can skip doing "Suchen" and manually enter the information.

If the issue is that after completing the first 3 form fields things do not work in the Panel (the first tab), could you also share a screenshot when OctoPod fails to connect?

Uwe-Kaki, 2020-11-30:
the issue is, that search ( "suchen" ) did not give me any results.

But : just tried again, got an update from OctoPrint for a new release, et voila : I can connect again.

But then, few minutes later, OcoPod App complaints about services on OctoPrint not running ... strange 

[image]

The message means : connection failed, service not running or port is wrong ( bzw : how good is your German ? )

in parallel testing the same movement of the z-axis works without issues with the alternative app from iPhone

gdombiak, 2020-11-30:
Summary of what I heard so far:
* "Suchen" does not find anything while the other app lists things
* Second tab "Bewegen" fails when you click on any of the buttons with a "connection failed" error

I'd assume that the first tab "Panel" is also failing with some kind of connection error. Based on all of this information, I wonder if you are using latest iOS version and denied local network access to OctoPod. This will not let "Suchen" work. In iOS, go to Settings -> OctoPod and verify that "Local Network" in enabled. I do not think you are using Cellular Data but you can also check that setting is also enabled.

gdombiak, 2020-11-30:
My German is zero. I had to google translate to say "Vielen Dank" :)

Uwe-Kaki, 2020-11-30:
Hi,
Local network access is allowed & yes, I am in my own network.

Your understanding is mostly correct, let me summarize :

Other APP
- the other app works all the time, no Problems ( I downloaded that App after the issues with Octopod)

Octopod App
-  ‚search‘ did not find any server, so no other panel worked
- got a new Octoprint release this evening, after that I could connect ( search showed a result )
- but few minutes later I get the connection errors ( service not running etc ) 
- from then on, no movement etc possible via Octopod

All the time the other app worked ( just to prove my Setup is ok )

I will call it a day for now, will check things tomorrow again

LG aus dem Norden / best regards

Uwe Knöfel

Uwe-Kaki, 2020-12-11:
Hi Gaston

 

Just to give some final feedback  : after OctoPrint got some urgency patches ( due to third party connecting issues J ) it is now all back to normal.

 

Sorry for the confusion and thx for quick responses

 

 

 

LG aus dem Norden

Uwe Knöfel