gdombiak/OctoPod #417: No camera feed on widget and Watch App

vordenken, 2020-12-02:
Hey,

since a few days ago I cannot get the camera feed to show up in the widget or the Apple Watch app. Works fine in the iOS app though.
I'm on iOS 14.2 and OctoPod v3.9 build 1. I don't really know if the problem appeared first with the latest OctoPod update.
If there is some kind of debug information I can give you let me know 👍 

Best regards,
VD

gdombiak, 2020-12-02:
Sorry to hear about that @vordenken. I'm curious when something stops working and the app was not updated. Have you updated OctoPrint recently? Have you changed the URL to the camera? Is the iOS app/iPhone at reach to the Apple Watch?

I do not have a way to access debug information so that makes it harder to troubleshoot. Could you try editing the widget and select (again) the printer and the camera? If you see a checkbox in both then chances are that that won't fix it but worth checking. That may fix the widget in case there was a change somewhere. Not sure about the apple watch unless there is an error when apple watch and iphone have an error talking. I fixed that in the coming release but not sure if that is your case.

Gaston

vordenken, 2020-12-02:
Thanks for your quick reply @gdombiak ! 
Actually after writing this post I checked the widget settings and apparently there wasn't a camera selected anymore. After selecting a camera again and closing the edit-menu the camera reappeared on the widget. So that's working now. 
Sadly this did not fix the watch camera feed. But no worries, the widget is my main way to take a quick glance anyways. 
How often does the camera picture on the widget get reloaded by the way?

Keep up your great work and this really awesome app!

Cheers,
VD

gdombiak, 2020-12-02:
That's great to hear! BTW, did both thing stopped working at the same time? 

iOS has a lot of restrictions on updating Widgets. We cannot do streaming or update very frequently. :( The best we can do is every 5 minutes. So OctoPod will refresh every 5 minutes when you are printing and every hour if you are not printing.

Thanks,
Gaston

gdombiak, 2020-12-02:
if you are using OctoPod plugin for OctoPrint then every time the plugin sends a push notification then the widget will be updated as well. Eg. when you start printing or pause or finish printing.

gdombiak, 2020-12-02:
I was thinking about something else to test. In the Apple Watch app, could you go to the first panel where you see list of printers and click on the refresh button? That will refresh data stored in the Apple Watch. I wonder if the URL to the camera got lost or something.

Let me know if that helped.
Gaston

vordenken, 2020-12-02:
I don't know if both stopped working at the same time unfortunately...
5 minutes is fine for me 👍🏼

I did press the refresh button on the first page but that didn't solve the problem. Also I noticed that the refresh button on the camera tab is kinda greyed out. Maybe that helps.
btw I use a standard usb webcam connected to a raspberry pi that runs octopi with no special config whatsoever. Octopod plug-in is installed as well.

Cheers,
VD

Gosarianer, 2020-12-07:
I have the same issue here. Honestly, I haven’t used the app for 2 month, because I hadn‘t much time to print and had to wait for a new hotend. Today I got a chinese webcam. I took my expensive Logitech webcam from my printer (need it for homework due to corona). After rebooting my raspberry pi, the new cam works good in Octopod App and Octoprint. But on Apple Watch, I have the same issue as vordenken. Refresh arrow not usable at camera tab on watch.
Regards,
André

vordenken, 2020-12-11:
So for some reason the camera view in the Apple Watch app started working again. I didn't change anything besides the ip from the octoprint host. Maybe something got reloaded because of this.

closed 2021-08-24 by vordenken