gdombiak/OctoPod #423: Tasmota broken

G3kid0, 2020-12-07:
Hi, with the new version 1.5.0 / 1.5.1 of Octoprint the Tasmota control is broken. 
Can you fix this ? 
Thank you

gdombiak, 2020-12-07:
Hi @G3kid0 , thanks for bringing this to my attention. Could you share more what is not working? Is it listing switches? getting their state? turning them on/off?

Thanks,
Gaston

G3kid0, 2020-12-07:
[image] 
Buttons are grey and can't click it. Of course it work on web interface. 
Maybe because of the new version of Octoprint-Tasmota plugin 

https://github.com/jneilliii/OctoPrint-Tasmota/commit/c86efa5856a397b609929339e6a84f1921a7376c

Thank you

gdombiak, 2020-12-07:
Thanks. That info helps. It seems like OctoPod is not able to know if the switches are on/off so it does not let you click on them. Will try it later tonight and see if I can add a fix for the next release

gdombiak, 2020-12-08:
Hi @G3kid0 , I do not have Tasmota plugs to fully reproduce your case. Could you log into OctoPrint and enable logging for Tasmota plugin using DEBUG level? Here is how I did it in my setup:
[image]

Once you did that:
1. go to OctoPod and go to the panel where you see your Tasmota plugs. 
1. Wait a couple of minutes
1. Go back to OctoPrint to Settings -> Logging and download the file **plugin_tasmota_debug.log**
1. Attach it here so I can see if the plugin is having any errors
1. You can remove the Logging Level that you added at the beginning if you want

I will keep testing locally in the meantime to see if I see something.

Thanks,
Gaston

gdombiak, 2020-12-08:
Hmm, could you check out also the log file **octoprint.log**? I have the following error in mine that is causing the plugs to not be refreshed. I wonder if you see a similar entry.

> 2020-12-07 20:00:03,549 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin tasmota
Traceback (most recent call last):
  File "/Users/gaston/repo/OctoPrintPython3/src/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/Users/gaston/repo/OctoPrintPython3/src/octoprint/util/__init__.py", line 1909, in wrapper
    return f(*args, **kwargs)
  File "/Users/gaston/repo/OctoPrint-Tasmota/octoprint_tasmota/__init__.py", line 501, in on_api_command
    return flask.jsonify(self.check_status("{ip}".format(**data), "{idx}".format(**data)))
  File "/Users/gaston/repo/OctoPrint-Tasmota/octoprint_tasmota/__init__.py", line 462, in check_status
    return {"currentState": "unknown", "ip": plugip, "idx": plugidx, "energy_data": energy_data, "sensor_data": sensor_data}
UnboundLocalError: local variable 'energy_data' referenced before assignment

gdombiak, 2020-12-08:
**Found the problem**. The plugin now works differently so OctoPod no longer works. I will contact the developer to see if he is going to also change all his other plugins to work like this one. Depending on his answer I will know what to fix (if need to do something, maybe he will rever that change).

G3kid0, 2020-12-08:
Thank you for the reactivity ! Do you still need the log files?

jneilliii, 2020-12-08:
I will look into putting the websocket communication layer back in later this week (maybe tomorrow after work), it's way too late for me right now (1:30AM). No changes on the OctoPod side should be necessary, but the Tasmota plugins will need to be upgraded once released.

gdombiak, 2020-12-08:
@jneilliii that's great to hear!!!! Thanks so much for being so accommodating.

gdombiak, 2020-12-08:
@G3kid0 no need for logs now. We got to the bottom of this. Thanks

jneilliii, 2020-12-09:
So since I don't have an apple device to test OctoPod and @gdombiak doesn't have a tasmota device to test the plugin I need someone to test to see if my 0.8.17rc1 version resolves this issue or not before I officially release. To test, if you are running OctoPrint 1.5.0+ you can change the release channel for the plugin in the software update settings by changing the Release Channel to `Release Candidate` otherwise you'll have to manually copy/paste the URL below into Plugin Manager > Get More > ...from URL and clicking Install. It should just install right over the existing version without any issues.

```
https://github.com/jneilliii/OctoPrint-Tasmota/archive/0.8.17rc1.zip
```

gdombiak, 2020-12-09:
Hi @jneilliii , I have the plugin installed in my dev environment and change the code that makes the actual call to the tasmota device to see how things work. I will give ti a try to the new version and let you know tonight how it worked in my local dev env.

G3kid0, 2020-12-11:
@jneilliii hmm it seems that it does not work. 
More test and details tonight after work.
Thanks for your work

gdombiak, 2020-12-11:
@jneilliii I posted in [this other issue](https://github.com/jneilliii/OctoPrint-Tasmota/issues/109#issuecomment-742948374) the possible cause why 0.8.17rc1 is still not working with OctoPod.

vordenken, 2020-12-11:
Wanted to let you guys know if you need help for testing this I could do it as well. I use OctoPod and tasmota switches.

dnasir, 2020-12-11:
@jneilliii I've tested the RC, and confirm that it did not solve the problem.

G3kid0, 2020-12-19:
Hi, no news about this ? :/

jneilliii, 2020-12-19:
Just now getting to the weekend, which is when I do most of my plugin work. Hopefully will have something new to test soon.

jneilliii, 2020-12-19:
I just put up an rc2 version if you guys could test and let me know. I basically added the websocket message back in the check status function @gdombiak, I hope that gets us back. Still unsure why adding them into the turn_on and turn_off commands wasn't working.

G3kid0, 2020-12-19:
@jneilliii Thank you, I tested the rc2, it seem to work. 
But now on Octopod there is an error message. @gdombiak  [image]

jneilliii, 2020-12-19:
you'll probably have to check the logs for errors. The only other change that may effect this is the permissions. I had to change the API permissions method due to future deprecation of the previous method. It should by default add the permissions to the admin group but if your API key is associated with a different user/group you'll have to adjust those permissions to allow control of tasmota.

[image]

Shaewen-Chronicles, 2020-12-22:
Hmm it changes the state of the switch but octopod shows the message @G3kid0 postet so i thing the problem is on the octopod side. If someone tells me what log files he need i can provide them

jneilliii, 2020-12-26:
I just pushed an update to the plugin that I think will allow OctoPod to work again. If you are already using the Release Candidate channel in Software Update for this plugin you should be able to just update to 0.8.17rc3. I misread what @gdombiak had told me in what OctoPod was expecting as a return code. I have now removed the 200 response codes so in theory should work the same as it did before the last update and give OctoPod what it expects to receive again without effecting the OctoPrint UI.

vordenken, 2020-12-26:
0.8.17rc4 works for me.

G3kid0, 2020-12-27:
@jneilliii yes 0.8.17rc4 works for me too.
Thanks