gdombiak/OctoPod #425: Feature request to have dashboard as a button

Jon-So, 2020-12-10:
Is there a way to have a "Dashboard" button to the right next to the printers dropdown instead of being in the dropdown list.

Jon-So, 2020-12-14:
[image]

Jon-So, 2020-12-14:
or have the printers removed for smaller screens and the 3 dashes instead and a Dashboard button.

Jon-So, 2020-12-14:
[image]

gdombiak, 2021-08-29:
Hey @Jon-So , it is clearly not a good UX what I did but if you click on the printer name you will go straight to the dashboard. I do not think I can add a button between the left button and the printer name.