gdombiak/OctoPod #426: Text color in Light Theme and Dark Mode

simonbogh, 2020-12-10:
Hi,

When the theme is set to Light, and your phone switches to Dark Mode, then the text you write to set the extruder and bed temperature becomes white text on white background. 
[image]
[image]

gdombiak, 2020-12-11:
Thanks so much @simonbogh for opening this issue. Themes have been around for a long long time and I had no idea we had this issue. :)

It is now fixed for the next release coming out in a few days.

Thanks,
Gaston

closed 2020-12-11 by gdombiak

simonbogh, 2020-12-13:
My pleasure, and happy to see the quick fix 😊

commit 77e5af54 (2020-12-11): Fixed background color when using Light theme and dark mode in iOS.

Closes #426