gdombiak/OctoPod #429: Disconnect cancels print

umtauscher, 2020-12-16:
Hi Gaston,
thank you very much for your app, I just installed it on iOS and tried it out - sadly during a print in progress.
I am a bit confused, because this issue is so old and still the disconnect stops the print and with that destroys the work in progress.
Just to clear things up "disconnect printer" would in my interpretation disconnect the app from the octoprint instance. Instead it disconnected the printer from Octoprint. Is that correct? I even enabled the read only mode but still it destroys the running print.
I would love to hear from you about this.
TIA
Umtauscher

gdombiak, 2020-12-16:
Hi @umtauscher ,

Thanks for the feedback. Disconnect will disconnect OctoPrint from the printer. If you are printing from OctoPrint then the print will be aborted. To prevent doing this by mistake, in OctoPod you can go to Settings -> Prompts and enable confirmation before disconnecting. I think I will add to the [FAQ](https://github.com/gdombiak/OctoPod/wiki/FAQ-using-the-app) what disconnect does so there is no (or less) confusion.

Thanks,
Gaston

gdombiak, 2020-12-16:
Added a new entry in the FAQ. Let me know if that helps and if you had other ideas.

Gaston

umtauscher, 2020-12-16:
Thanks Gaston for your quick answer.
I know now, what this does and I personally find it to really be in the wrong spot. Disconnecting will always lead to a print failure and I think this should be in the administrative spot where the other functions like restarting octoprint is.
I find this function should not be on such a prominent spot on the main screen.
In fact, who needs this function anyway while remotely checking a print?
It would be very nice, if you would reconsider. Even when you set the UI to read only this function is active. That makes the readonly setting obsolete in my opinion.
Anyways, thanks for your hard work.
Cheers
Umtauscher

gdombiak, 2020-12-16:
That's a good point! I'm not sure about reorganizing the UI at this point but I can easily make the button respect the read-only setting. If many other people feel the same way about where the button is placed then we can figure out how to reorganize the UI.

umtauscher, 2020-12-16:
Super, thanks. 
Very nice of you!
Cheers

gdombiak, 2021-01-31:
@umtauscher I was about to implement this and noticed that it is already implemented. Disconnect button is disabled when app is locked.

umtauscher, 2021-01-31:
Hi,

if you think you implemented it, then it simply doesn‘t work.
I just tested:
I activated „automatically lock during print“
Started a print
Everything is disabled only „disconnect“ is still active.

But I cannot pause a print, that‘s also disabled.
I cannot cancel the print - ok that may be as expected.
I cannot disable the lock function during print.

I hope that helps to clarify the problem.
Thanks

mindnever, 2021-05-22:
Hi,

How about having the essential info included in confirmation request itself as a first step. I also like the idea of (re)moving this button completely from such important part of the screen. How often is this functionality even used?

Thanks

umtauscher, 2021-05-22:
removing it from the main screen would be fine by me.
I would never disconnect from a remote printer anyways.
Thanks

Wilhelm

Xalior, 2021-08-29:
Can confirm this just happened to me 3hrs into an 11hr job! 😅 

-Dx

gdombiak, 2021-08-30:
Hey @Xalior, under Settings -> Prompts you can enable/disable a prompt before connecting or before disconnecting. There is no connection to connect/disconnect between OctoPod and OctoPrint. These actions are between OctoPrint and the Printer.