gdombiak/OctoPod #430: Please i need PSU Plugin into the app

fadi9696, 2020-12-17:
Please i need PSU Plugin into the app to turn printer off and on

gdombiak, 2020-12-17:
Hi @fadi9696 , here is the list of [supported plugins](https://github.com/gdombiak/OctoPod/wiki/Supported-Plugins). PSU Control is supported. Are you asking for that plugin?

fadi9696, 2020-12-17:
Yes i'm asking for PSU plugin its on the list but how to show it on app?

gdombiak, 2020-12-17:
The steps would be:
1. Install [PSU Control](https://plugins.octoprint.org/plugins/psucontrol/) plugin in OctoPrint
2. Open OctoPod
3. In the main window (Panel) you can swipe from right to left and you will see more subpanes. You will see the temp graph pane and as you swipe more then you will see the PSU Control pane

Hope that helps,
Gaston

fadi9696, 2020-12-17:
Thank you so much you are awesome

gdombiak, 2020-12-17:
Let me know if we can close this ticket now. BTW, all other plugins work in a similar way. You swipe and you access more and more functionalities

gdombiak, 2020-12-17:
One last ask: Please leave your app review if you haven't given one already. This helps spread the word and help other people find the app. Thanks

fadi9696, 2020-12-17:
I totaly did'nt knew about this awesome feature, i will leave a review about it, thank you again

gdombiak, 2020-12-22:
Closing the ticket for now. Let me know if you need further assistance.

Gaston

closed 2020-12-22 by gdombiak