gdombiak/OctoPod #433: Add support for RTSP stream from IP cameras

tideline3d, 2020-12-19:
Have you considered supporting RTSP streams from IP cameras?   I currently have Unifi Protect Flex cameras monitoring my printers.  They provide a snap.jpeg and an RTSP stream. 

It'd be sweet to use that RTSP stream in the app since browsers don't really support that. 

Ideally the main page would just refresh the snap jpeg occasionally then when you tap on the image, have that load up the stream. 

Currently Octopod doesn't seem to support a snap.jpeg URL as the camera URL.  I just get a gray box for the camera.

gdombiak, 2020-12-19:
Hi @tideline3d , I recently added support for HLS streaming. You could convert [RTSP in HLS](https://medium.com/androvideo/convert-rtsp-to-hls-using-ffmpeg-2fe2cdf3a0de) and watch your cameras from octopod. You’ll even be able to do PIP and other cool things with HLS. Let me know how it goes. 

Thanks 
Gaston

tideline3d, 2020-12-20:
@gdombiak Thanks, I got that setup with a couple Restreamer docker instances, seems to work great!

closed 2020-12-20 by tideline3d

gdombiak, 2020-12-22:
Hey @tideline3d, that is super great to hear. I think that many other people would be interested in doing the same thing. Did you follow any guide? I can create a wiki page to document this or let me know if you would like to do this? I can promote it our the facebook group to spread the word.

Thanks,
Gaston

tideline3d, 2020-12-22:
@gdombiak I can write up a quick guide if you'd like.  Have any preference for where I should put it in the Wiki?

gdombiak, 2020-12-23:
That would be fantastic. I created this [wiki page](https://github.com/gdombiak/OctoPod/wiki/How-to-watch-RTSP-streams-from-IP-cameras%3F) that you should be able to edit. Let me know once you are done and I will share it with the community. 

Gaston