gdombiak/OctoPod #435: Camera Connection Failed

billyjacoby, 2020-12-23:
For about a week now I've been getting a "Connection Failed" error when trying to view the camera of my printer.

This feature has been working for the entire time that I've been using Octopod before, and I haven't changed anything on my end to provoke this change.

I've deleted and re-added the printer configuration but this hasn't fixed the issue.

I downloaded the app from the app store rather than building from source, so if it would help I could try compiling and debugging through Xcode - unless there's a simpler way.

gdombiak, 2020-12-24:
Hi @billyjacoby, could you click on the link shown in OctoPod? That will open the camera on safari using the same URL. Let me know what happens when you click. If Safari has the same error AND the URL is correct then this is an OctoPi configuration or network issue. You can also share the URL here in case that gives us some info.

Thanks,
Gaston

billyjacoby, 2020-12-24:
The camera can be viewed without issue using Safari. I am also still receiving images in my notifications as well.