gdombiak/OctoPod #439: (Mexican Spanish Siri) Siri wont show or speak time left.

EjianYT, 2021-01-03:
I was on iOS 13, and that was working ok... buy a new iphone (11) update old iphone (7) to iOS 14.3 and the shortcut to time left wont work anymore, just say "ok mostrando con octopod" (ok showing with octopod)... and wont display any numbers on the notification or say anithing else... in english it does say the time (a little weird) but dosent show timremaining on notification either.

EjianYT, 2021-01-05:
Hola, súper que hablas español... te hice un video del funcionamiento de Siri, planeó hacer un video de octopod en mi canal, y me gustaría que funcione perfectamente para el video. 

Muchas gracias, voy a donar algo al proyecto 😊

https://youtu.be/jzQpJAhYaC8

gdombiak, 2021-01-10:
@EjianYT , aca veo que aparece el tiempo restante https://youtu.be/jzQpJAhYaC8?t=23. Es en el segundo 23. Es eso lo que pedis?

gdombiak, 2021-01-17:
@EjianYT alguna novedad con mi ultima respuesta?

Saludos,
Gaston

EjianYT, 2021-01-17:
Hola, perdón me perdí con tanta notificación... si sale el tiempo ahí en ese segundo, pero solo es cuando vas a shortcuts y lo activas desde la app (y Siri no habla ahí)... si usas a Siri con comando de voz, no sale nada. 
Adicionalmente lo probé en el iPhone de una amiga (8) y también no funciona el que de el tiempo restante de la impresión. No he podido conseguir un iPhone con un iOS anterior para probar ahí.

gdombiak, 2021-01-17:
Me suena a que es un error de iOS lamentablemente. Desde OctoPod todo lo que puedo hacer es devolver la information. El resto depende de Siri para que lo muestre. :( No es la mejor respuesta pero creo que esta afuera de mi alcance arreglar esto.

Gaston