gdombiak/OctoPod #440: Add custom default temperatures

MunchHansen, 2021-01-05:
It would be great - if it was possible to change the default temperatures shopping jo when setting the temperature for either the end or bed.

gdombiak, 2021-01-05:
Hi @MunchHansen,

OctoPod should render the temperatures that you defined in OctoPrint. Have you set them here?
[image]

Regards,
Gaston

MunchHansen, 2021-01-05:
Hi Gdombiak.

My bad sorry and thanks for showing me! I had not discovered that option!

Have a nice day.

gdombiak, 2021-01-05:
Cool. I will add it to the [FAQ](https://github.com/gdombiak/OctoPod/wiki/FAQ-using-the-app). I'm sure other people have the same question.

closed 2021-01-05 by gdombiak