gdombiak/OctoPod #441: Compatibilidad con octolapse

EjianYT, 2021-01-05:
Hola de nuevo, y gracias de nuevo por tu asombrosa app (para mi la mejor y más completa qué hay).

Hay un problema con octolapse, si octolapse está activo, y se inicia una impresión desde octopod, parece que no recibe la instrucción de empezar a imprimir... ya que octolapse pide confirmación del time lapse que va a hacer, y hay que entrar a octoprint para poder aceptar el mensaje de octolapse.

Saludos.

gdombiak, 2021-01-10:
Hola @EjianYT,

Te referis a estoa confirmacion? Esto aparece cada vez que empezas una impresion?
[image]

Gracias,
Gaston

EjianYT, 2021-01-10:
Si esa, yo uso Cura conectado a octoprint, y también tu app, y sale la confirmación, y no deja empezar la impresión.

gdombiak, 2021-01-10:
Genial! Gracias por la confirmacion de lo que pedis. Mientras lo "arreglo", podes desactivar la confirmacion asi ese dialogo no aparece. 

[image]

EjianYT, 2021-01-11:
Listo, puse la opción de que se quitara automáticamente después de un tiempo, por que si necesito que me acuerde que está activo.