gdombiak/OctoPod #443: Allow printer selection by swiping from ALL pages

iwalldock, 2021-01-12:
I have a small print farm and I use this app often. I love being able to switch between printers by swiping on the printer title at the top of the app on the Home Screen. But I hate it when I’m on a different page (like movement, or temp graph, etc) and for some reason this swipe feature doesn’t work. Please make it so we can swipe between printers whenever the title is shown at the top for a more uniform user experience 👍🏼

gdombiak, 2021-01-17:
Hi @iwalldock, are you not able to use the swipe to change printers when you are in the **Panel** tab? I'm not able to reproduce this so hard to fix. I can go to the temp chart or any other subpanel when in the Panel tab and still can swipe to change between printers.

Do you think you could share some steps so I can reproduce this so I can fix it? BTW, can you click on the title to open the dashboard when swipe is not working? Does pressing on **Printers** work when swipe stops working?

Thanks,
Gaston

iwalldock, 2021-01-17:
I’m sorry, maybe I didn’t describe the request well enough.  I CAN swipe between printers on the Panel tab.  I cannot switch printers from any other tab.  My request would be to swipe on any tab/page (or pull up printer selection screen by tapping).  The thought is, when I want to print new files on all ten printers, I have to go to the files tab, choose a file, then unnecessarily go back to the panel tab, swipe for the next printer, then back to the files tab, select a new file, etc.  If you could swipe between printers on ANY tab, then this process would be so much more efficient and intuitive.

Thanks!

George Hires
Founder & CEO
www.iWalldock.com <http://www.iwalldock.com/>

Follow us on Facebook <http://www.facebook.com/iwalldock>

gdombiak, 2021-01-17:
Gotcha! Makes sense. The request is to have the swipe available when in the **Move**, **Files** and **Terminal** tab. A few times I had the same need and after swiping nothing happened .... :(

Will add this to a future release.

Thanks,
Gaston

iwalldock, 2021-01-17:
Perfect!  Thank you!!

George Hires
Founder & CEO
www.iWalldock.com <http://www.iwalldock.com/>

Follow us on Facebook <http://www.facebook.com/iwalldock>

commit 8e86af75 (2021-06-26): Allow printer selection by swiping from ALL pages

Closes #443
closed 2021-06-26 by gdombiak