gdombiak/OctoPod #444: Apple Watch App Not Functioning

johnmiddleton12, 2021-01-13:
Hello, I'm using OctoPod with my phone and installed the app on my watch, but when using it I don't see any of the printer information, just the name of the printer, and the second tab is blank except the icons. The camera tab is blank as well.
It was initially only saying to open the OctoPod app on my phone to get the printer settings, but that seemed to only sync the name of the printer. Thank you!
Watch OS version - 7.2
Series 4 GPS

gdombiak, 2021-01-17:
Hi @johnmiddleton12, so after opening the iOS app you got the list of printers in the Apple Watch but no information about the printer status or current print or webcam image appear?

Could you share a screenshot of your watch? I'm assuming that you are printing and see no info. If you are not printing then second page will just show printer status.

Hard to tell what is going on with this little info. Not that we can get much more since I cannot debug your watch. :( I wonder if there is a connectivity issue between the watch and the phone. Could you try rebooting for the watch and the phone and see if things work after that? 

Thanks,
Gaston

brs165, 2021-03-22:
I believe I had a similar issues where do data was being passed between the phone/watch.  I tried reboot both with no resolution.  I uninstalled the app from the watch, and then reinstalled it to the watch.  That resolved the issue for me.