gdombiak/OctoPod #445: Need support for client certificate authentication

DBa2016, 2021-01-13:
My Octoprint instance is protected by client certificates (since I am too lazy to enter passwords), and my iPhone and iPad have the correct certificates, so passwordless login to Octoprint Web works fine. Is there any way to make Octopod use these?

gdombiak, 2021-01-13:
Hi @DBa2016 , very interesting request. How do you configure the client request in iOS? I never used that feature. Happy to add this feature once I know how to use it.

DBa2016, 2021-01-14:
Uhhh...I have no idea how the "client request" is configured... For the Web access, I imported the .p12 file into the central store and Safari automagically picks it up when accessing my OctoPrint instance (the client certificate is enforced by HAProxy).

This way, I don't need to enter the password on the HTTP stage (still need to remove password in Octoprint itself once my current print finishes), just confirm "yes, I want to use this client certificate".

I wouldn't make it the only option (creating client certificates needs some infrastructure...), but having it as a choice would be nice.

DBa2016, 2021-01-15:
I don't have any iOS programming experience, but this is what I found googling so far:
https://stackoverflow.com/questions/55758377/how-can-i-send-https-request-with-p12-certificate-on-objective-c-or-swift
https://blog.xdumaine.com/authenticating-an-http-request-with-client-certificates-in-ios

For now, I will probably define an extra frontend in HA, just for Octopod :)